gabriel-milan/TinyMPU6050 #1: Examples?

feldon30, 2019-03-15:
Thank you for posting this. However I find no Arduino or AVR code examples on the use of this library.

gabriel-milan, 2019-03-19:
Hi @feldon30, thanks for posting!

You can find a basic example on how showing angles here: https://github.com/gabriel-milan/TinyMPU6050/tree/master/examples/PlatformIO_Angles_Example

It's a PlatformIO (https://platformio.org/) project, as I use VSCode for coding, but if you just copy/paste the code at https://github.com/gabriel-milan/TinyMPU6050/blob/master/examples/PlatformIO_Angles_Example/src/main.cpp on your Arduino IDE or whatever, it should work.

Anyway, if you need a specific kind of example code, just let me know and I'll post it ASAP.

feldon30, 2019-03-19:
Thank you kindly.

closed 2019-03-19 by gabriel-milan