gabriel-milan/TinyMPU6050 #4: Getting missing AccZ GyY values from calibration

foxabilo, 2019-03-28:
Getting this from Calibration mode 
13:58:48.156 -> 88 loops / 4 axis calibrated. Missing:  AccZ GyY
13:58:48.361 -> 89 loops / 4 axis calibrated. Missing:  AccZ GyY
...

Nope I am not on a roller coaster.

gabriel-milan, 2019-03-28:
Hi @foxabilo, thanks for posting!

As I've realized, in some architectures calibration may not converge (in my ESP32, for example) with the default deadzones. As a workaround, you can set new deadzone values and check if it'll converge.
For my ESP32 I'll be using 0.01 m/s² for accelerometer and 0.06 degrees/second for the gyro (defaults are 0.002 m/s² and 0.015 degrees/second, respectively).

Let me know if this helps!

foxabilo, 2019-03-28:
Ahh! Yes I am running on a WiFi KIT 32 ESP32 with the little oled from HELTECH, I'll try the new values and see how it does, I kind of need calibration to be as good as it can get and then just run with what its got as having the calibration cycle potentially stop a remote unit from booting and doings its initialisation functions would be undesirable.

foxabilo, 2019-03-28:
Did you mean to half the values? that would make it more sensitive I think, I have changed the values to 0.04m/ss and 0.03d/s and it now goes up to 5 axis calibrated, but the Z remains stubborn, I will see at what value it passes.

foxabilo, 2019-03-28:
`#define DEFAULT_ACCEL_DEADZONE          0.011 // m/s²
#define DEFAULT_GYRO_DEADZONE           0.021 // Degrees/second`
Those are the absolute minimum values I can use for it to pass calibration

gabriel-milan, 2019-03-28:
> Those are the absolute minimum values I can use for it to pass calibration

I'm glad you found some acceptable values for your application!

Yet, if you need to change these values later, you don't have to actually look for it on the header file. There are methods for that, as below:
```
void SetAccelDeadzone (float deadzone)
void SetGyroDeadzone (float deadzone)
```
Results will be the same, but it makes your code more intuitive.

> Most of the issue is on the Z axis

Yeah, I realize that. The reason for this is that calibration for the Z axis is quite different from the others because of the offset (gravity). I had a hard time while trying to make Z axis to work and probably there's a better way of doing this but it works!

foxabilo, 2019-03-28:
Indeed, perhaps a completely separate Z deadband?

gabriel-milan, 2019-03-28:
> perhaps a completely separate Z deadband?

Yeah idk, maybe as something optional. Great idea, I'll consider implementing this on the next releases. Thank you for the feedback!

If you need help with anything else, let me know!

closed 2019-03-28 by gabriel-milan