gabriel-milan/TinyMPU6050 #7: Updating to V0.4.3

foxabilo, 2019-03-30:
After using the build in Arduino IDE update library function to V 0.4.3 the following is received in the debug window

`Invalid library found in C:\Users\<user name>\OneDrive\Documents\Arduino\libraries\TinyMPU6050: no headers files (.h) found in C:\Users\<user name>\OneDrive\Documents\Arduino\libraries\TinyMPU6050`

This may be a OneDrive path issue, but deleting the library manually and then manually putting the downloaded extracted zip version from GitHub fixes the issue.

gabriel-milan, 2019-04-02:
Hi @foxabilo!

This is probably a path issue, something that's not uncommon on Windows machines. My personal recommendation is, if you really want to develop on Windows, that you use VSCode (https://code.visualstudio.com/) + PlatformIO extension for VSCode (https://platformio.org/) on your machine.

closed 2019-04-02 by gabriel-milan

foxabilo, 2019-04-02:
Owww, that looks shiny... I am ripping my hair out with Arduino IDE, it reminds me or borland turbo pascal with the DOS interface