gabriel-milan/TinyMPU6050 #8: ESP32 with MPU6050 ping no work

cruizg, 2019-04-15:
can you helo me, i have a ESP32 WEMOS OLED , i trying your library but dont work , only print constants values :

> 9:45:02.820 -> GyroY = -0.02 degrees/second
09:45:02.820 -> GyroZ = -0.02 degrees/second
09:45:02.820 -> --- Accel angles:
09:45:02.820 -> AccelAngX = 180.00
09:45:02.867 -> AccelAngY = 180.00
09:45:02.867 -> AccelAngZ = 45.00
09:45:02.867 -> --- Gyro angles:
09:45:02.867 -> GyroAngX = -5.45
09:45:02.867 -> GyroAngY = -5.45
09:45:02.867 -> GyroAngZ = -5.45
09:45:02.867 -> --- Filtered angles:
09:45:02.867 -> FilteredAngX = 177.94
09:45:02.867 -> FilteredAngY = 177.94
09:45:02.867 -> FilteredAngZ = 42.94
09:45:02.867 -> --- Angle filter coefficients:
09:45:02.867 -> Accelerometer percentage = 0.02%
09:45:02.867 -> Gyroscope percentage = 0.98%
09:45:02.867 -> --- Deadzone:
09:45:02.867 -> Accelerometer deadzone = 0.00 m/s²
09:45:02.867 -> Gyroscope deadzone = 0.01 degrees/second

cruizg, 2019-04-15:
i was testing disconect MPU to ESP32 but the result is the same

gabriel-milan, 2020-03-10:
Hi @cruizg!

I'm back on developing this library, so your issue can probably be solved on release 0.5.0, if still in need. Let me know as soon as I release it!

gabriel-milan, 2020-03-10:
Release 0.5.0 is now out [here](https://github.com/gabriel-milan/TinyMPU6050/releases/tag/0.5.0).

I'll close this for now. If you continue to have issues please contact. Thanks!

closed 2020-03-10 by gabriel-milan