gabriel-milan/TinyMPU6050 #9: Issue with calibration

st4s1k, 2019-04-19:
12:58:21.597 206 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:21.775 207 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:21.967 208 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:22.161 209 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:22.365 210 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:22.584 211 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:22.815 212 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:23.012 213 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:23.141 214 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:23.338 215 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:23.550 216 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:23.754 217 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:23.931 218 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:24.139 219 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:24.367 220 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:24.577 221 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:24.704 222 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:24.901 223 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:25.189 224 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:25.294 225 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:25.500 226 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:25.711 227 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:25.881 228 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:26.076 229 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:26.273 230 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:26.471 231 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:26.685 232 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:26.873 233 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:27.085 234 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:27.297 235 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:27.444 236 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:27.709 237 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:27.837 238 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:28.047 239 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:28.244 240 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:28.464 241 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:28.667 242 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:28.864 243 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:29.013 244 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:29.206 245 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:29.409 246 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:29.594 247 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:29.826 248 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:29.992 249 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:30.182 250 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:30.377 251 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:30.572 252 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:30.771 253 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:30.965 254 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:31.168 255 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:31.398 256 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:31.614 257 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:31.811 258 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:31.941 259 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:32.138 260 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:32.363 261 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:32.527 262 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:32.771 263 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:32.918 264 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:33.116 265 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:33.311 266 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:33.507 267 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:33.733 268 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:33.935 269 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:34.135 270 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:34.324 271 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:34.535 272 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:34.693 273 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:34.876 274 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:35.074 275 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:35.266 276 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:35.490 277 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:35.677 278 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:35.913 279 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:36.051 280 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:36.288 281 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:36.442 282 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:36.664 283 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:36.839 284 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:37.049 285 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:37.299 286 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:37.451 287 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:37.617 288 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:37.806 289 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:38.003 290 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:38.204 291 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:38.402 292 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:38.617 293 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:38.815 294 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:38.981 295 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:39.176 296 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:39.374 297 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:39.589 298 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:39.761 299 loops / 4 axis calibrated. Missing:  AccZ GyY
12:58:40.002 300 loops / 4 axis calibrated. Missing:  AccZ GyY

OferShouval, 2019-04-24:
having the same issue - 

=====================================
12:19:55.970 -> Starting calibration...=====================================
12:19:57.605 -> Starting calibration...
1 loops / 2 axis calibrated. Missing: AccX AccZ GyX GyY
2 loops / 3 axis calibrated. Missing: AccX AccZ GyX
3 loops / 3 axis calibrated. Missing: AccX AccZ GyX
4 loops / 3 axis calibrated. Missing: AccX AccZ GyX
5 loops / 3 axis calibrated. Missing: AccX AccZ GyX
6 loops / 3 axis calibrated. Missing: AccX AccZ GyX
7 loops / 3 axis calibrated. Missing: AccX AccZ GyX
8 loops / 3 axis calibrated. Missing: AccX AccZ GyX
9 loops / 3 axis calibrated. Missing: AccX AccZ GyX
10 loops / 3 axis calibrated. Missing: AccX AccZ GyX
11 loops / 3 axis calibrated. Missing: AccX AccZ GyX
12 loops / 3 axis calibrated. Missing: AccX AccZ GyX
13 loops / 3 axis calibrated. Missing: AccX AccZ GyX
14 loops / 3 axis calibrated. Missing: AccX AccZ GyX
15 loops / 3 axis calibrated. Missing: AccX AccZ GyX
16 loops / 3 axis calibrated. Missing: AccX AccZ GyX
17 loops / 3 axis calibrated. Missing: AccX AccZ GyX
18 loops / 3 axis calibrated. Missing: AccX AccZ GyX
19 loops / 3 axis calibrated. Missing: AccX AccZ GyX
20 loops / 3 axis calibrated. Missing: AccX AccZ GyX
21 loops / 3 axis calibrated. Missing: AccX AccZ GyX
22 loops / 3 axis calibrated. Missing: AccX AccZ GyX
23 loops / 3 axis calibrated. Missing: AccX AccZ GyX
24 loops / 3 axis calibrated. Missing: AccX AccZ GyX
25 loops / 3 axis calibrated. Missing: AccX AccZ GyX
26 loops / 3 axis calibrated. Missing: AccX AccZ GyX
27 loops / 3 axis calibrated. Missing: AccX AccZ GyX
28 loops / 3 axis calibrated. Missing: AccX AccZ GyX
29 loops / 3 axis calibrated. Missing: AccX AccZ GyX
30 loops / 3 axis calibrated. Missing: AccX AccZ GyX
31 loops / 3 axis calibrated. Missing: AccX AccZ GyX
32 loops / 3 axis calibrated. Missing: AccX AccZ GyX
33 loops / 3 axis calibrated. Missing: AccX AccZ GyX
34 loops / 3 axis calibrated. Missing: AccX AccZ GyX
35 loops / 3 axis calibrated. Missing: AccX AccZ GyX
36 loops / 3 axis calibrated. Missing: AccX AccZ GyX
37 loops / 3 axis calibrated. Missing: AccX AccZ GyX
38 loops / 3 axis calibrated. Missing: AccX AccZ GyX
39 loops / 3 axis calibrated. Missing: AccX AccZ GyX
40 loops / 3 axis calibrated. Missing: AccX AccZ GyX
41 loops / 2 axis calibrated. Missing: AccX AccZ GyX GyY
42 loops / 3 axis calibrated. Missing: AccX AccZ GyX
43 loops / 3 axis calibrated. Missing: AccX AccZ GyX
44 loops / 3 axis calibrated. Missing: AccX AccZ GyX
45 loops / 2 axis calibrated. Missing: AccX AccZ GyX GyY
46 loops / 2 axis calibrated. Missing: AccX AccZ GyX GyY
47 loops / 3 axis calibrated. Missing: AccX AccZ GyX
48 loops / 3 axis calibrated. Missing: AccX AccZ GyX
49 loops / 3 axis calibrated. Missing: AccX AccZ GyX
50 loops / 3 axis calibrated. Missing: AccX AccZ GyX
51 loops / 3 axis calibrated. Missing: AccX AccZ GyX
52 loops / 3 axis calibrated. Missing: AccX AccZ GyX
53 loops / 3 axis calibrated. Missing: AccX AccZ GyX
54 loops / 3 axis calibrated. Missing: AccX AccZ GyX
55 loops / 3 axis calibrated. Missing: AccX AccZ GyX
56 loops / 3 axis calibrated. Missing: AccX AccZ GyX
57 loops / 3 axis calibrated. Missing: AccX AccZ GyX
58 loops / 3 axis calibrated. Missing: AccX AccZ GyX
59 loops / 3 axis calibrated. Missing: AccX AccZ GyX
60 loops / 3 axis calibrated. Missing: AccX AccZ GyX
61 loops / 3 axis calibrated. Missing: AccX AccZ GyX
62 loops / 3 axis calibrated. Missing: AccX AccZ GyX
63 loops / 3 axis calibrated. Missing: AccX AccZ GyX
64 loops / 3 axis calibrated. Missing: AccX AccZ GyX
65 loops / 3 axis calibrated. Missing: AccX AccZ GyX
[... 510 lines cut ...]
276 loops / 3 axis calibrated. Missing: AccX AccZ GyX
277 loops / 3 axis calibrated. Missing: AccX AccZ GyX
278 loops / 3 axis calibrated. Missing: AccX AccZ GyX
279 loops / 3 axis calibrated. Missing: AccX AccZ GyX
280 loops / 3 axis calibrated. Missing: AccX AccZ GyX
281 loops / 3 axis calibrated. Missing: AccX AccZ GyX
282 loops / 3 axis calibrated. Missing: AccX AccZ GyX
283 loops / 3 axis calibrated. Missing: AccX AccZ GyX
284 loops / 3 axis calibrated. Missing: AccX AccZ GyX
285 loops / 3 axis calibrated. Missing: AccX AccZ GyX
286 loops / 3 axis calibrated. Missing: AccX AccZ GyX
287 loops / 3 axis calibrated. Missing: AccX AccZ GyX
288 loops / 3 axis calibrated. Missing: AccX AccZ GyX
289 loops / 3 axis calibrated. Missing: AccX AccZ GyX
290 loops / 3 axis calibrated. Missing: AccX AccZ GyX
291 loops / 3 axis calibrated. Missing: AccX AccZ GyX
292 loops / 3 axis calibrated. Missing: AccX AccZ GyX
293 loops / 3 axis calibrated. Missing: AccX AccZ GyX
294 loops / 3 axis calibrated. Missing: AccX AccZ GyX
295 loops / 3 axis calibrated. Missing: AccX AccZ GyX
296 loops / 3 axis calibrated. Missing: AccX AccZ GyX
297 loops / 3 axis calibrated. Missing: AccX AccZ GyX
298 loops / 3 axis calibrated. Missing: AccX AccZ GyX
299 loops / 3 axis calibrated. Missing: AccX AccZ GyX
300 loops / 3 axis calibrated. Missing: AccX AccZ GyX

sorgx, 2019-05-01:
> having the same issue -
>
> =====================================
> 12:19:55.970 -> Starting calibration...=====================================
> 12:19:57.605 -> Starting calibration...
> 1 loops / 2 axis [...]

I'm having the exact problem. Too bad it's no working yet. The other libs are way more complex.

TAz00, 2019-07-31:
But it does use 0x68

/*
 *	Default MPU6050 address
 */
#define MPU6050_ADDRESS     	0x68

I am also having the same problem.

AlexFundorin, 2019-08-28:
+1 here. In fact, I didn't test the console output. It's just that I can see measurements (drifting) from the module on the oled display, when calibrate function is disabled, and when it's enabled, the display won't even start, meaning that some endless loop is happening.

gabriel-milan, 2020-03-10:
Hi everybody.

Calibration has always been an issue with this library. I'm implementing a new algorithm for that and probably on release 0.5.0 your issues will be solved. Please let me know as soon as I release it.

gabriel-milan, 2020-03-10:
Release 0.5.0 is now out [here](https://github.com/gabriel-milan/TinyMPU6050/releases/tag/0.5.0).

I'll close this for now. If you continue to have issues please contact. Thanks!

closed 2020-03-10 by gabriel-milan