gabriel-milan/TinyMPU6050 #12: Unreliable endianness of retrieved data

edgar-bonet, 2020-03-06:
The method `MPU6050::UpdateRawAccel()` contains [the following line][accel]:

```c++
rawAccX = wire->read() << 8 | wire->read();
```

and similar lines for `rawAccY` and `rawAccZ`. The same idiom [is used in `MPU6050::UpdateRawGyro()`][gyro].

This kind of expression should be avoided, as it does not guarantee the endianness of the retrieved data. This is because, in C++, [the order of evaluation of the arguments of the `|` operator is unspecified][cppref].

Presumably, the current version of gcc, when invoked with the options passed by the current version of the Arduino IDE, generates assembly that orders the bytes in the expected way. But this cannot be relied upon, and can very well depend on the compiler, compiler version, compiler options, or even surrounding code. This is the kind of software error that cannot be unveiled by testing, as the code can “just work” even if it is incorrect.

The simplest solution is to issue the calls to `wire->read()` within separate statements. Unlike the arguments of the `|` operator, separate statements have a well define order of evaluation:

```c++
rawAccX = wire->read() << 8;
rawAccX |= wire->read();
```

See also [this post on the Arduino forum][post].

[accel]: ../blob/0.4.3/src/TinyMPU6050.cpp#L113
[gyro]: ../blob/0.4.3/src/TinyMPU6050.cpp#L134-L136
[cppref]: https://en.cppreference.com/w/cpp/language/eval_order
[post]: https://forum.arduino.cc/index.php?topic=659333.msg4443224#msg4443224

gabriel-milan, 2020-03-10:
Thanks for the advice!
I had abandoned this library for a good while and now I see that I shouldn't.
I'm getting back to it and your suggestion is already on the way also.

edgar-bonet, 2020-03-10:
You missed [this one][]:

```c++
rawGyroZ = wire->read() << 8 | wire->read();
```

[this one]: ../blob/5264a0dad3928230909409999ad4b670e8cba899/src/TinyMPU6050.cpp#L158

PR #13 (2020-03-11): Ensure reliable endianness of retrieved data
In the expression

```c++
wire->read() << 8 | wire->read()
```

C++ does not guarantee the order of evaluation of the two calls to `wire->read()`. We get this guarantee by putting the calls into separate statements.

Fixes #12.
commit 6726bce1 (2020-03-11): Ensure reliable endianness of retrieved data

In the expression

	wire->read() << 8 | wire->read()

C++ does not guarantee the order of evaluation of the two calls to
wire->read(). We get this guarantee by putting the calls into separate
statements.

Fixes #12.
closed 2020-03-11 by gabriel-milan