gabriel-milan/TinyMPU6050 #18: Missing yaw compensation for pitch and roll gyro angles

bart, 2020-04-23:
Hey, first of all I want to thank you for the amazing work. During diving into this entire topic I came up with lots of libraries and I really like yours a lot. I just have a short question about yaw compensation like described here: https://youtu.be/4BoIE8YQwM8?t=606

Is it possible that this part is missing in your library?

Thanks! bart

gabriel-milan, 2020-04-24:
Hi @bart, sorry for late answer.

Thank you for the nice words and thank you again for bringing improvement to this library.

This compensation is not yet implemented here, but it surely can be. I just wonder if, on implementation, this compensation should be updated simultaneously because, on the video, he shows this

```ino
angle_roll -= angle_pitch * sin(gyro_z * 0.000001066);
angle_pitch += angle_roll * sin(gyro_z * 0.000001066);
```

but this is not being updated simultaneously, as `angle_roll` and `angle_pitch` become dependent of each other. For simultaneous update, it would be something like this:

```ino
new_angle_roll = angle_pitch - angle_pitch * sin(gyro_z * 0.000001066);
new_angle_pitch = angle_roll + angle_roll * sin(gyro_z * 0.000001066);
angle_roll = new_angle_roll;
angle_pitch = new_angle_pitch;
```

(maybe there's a better/faster way to implement this, but I just wanted to give you the idea)

Would you know if it should be updated simultaneously?
And is it of your interest to implement this improvement and opening a PR?

bart, 2020-04-24:
Hey, thanks for your quick response and especially thanks for the remark in terms of simultaneous updates. It makes more sense for me to store data first in variables to prevent them from affecting each other. By the way the `0.000001066` factor must also be replaced by the actual time passed because this is just true for 250Hz updates. I currently implemented my own calculations but can create a PR later to add this feature to your library as well, yeah!

bart, 2020-04-24:
By the way the equation must be this instead, right?

```
new_angle_roll = angle_roll - angle_pitch * sin(gyro_z * 0.000001066);
new_angle_pitch = angle_pitch + angle_roll * sin(gyro_z * 0.000001066);
```

gabriel-milan, 2020-04-24:
On the video you sent you can see [here](https://youtu.be/4BoIE8YQwM8?t=728) the equations I wrote just above. I can't tell whether they're correct, unfortunately. 

Anyway, as you showed interest on writing a PR, I'll eagerly wait for it.
If you need help on anything let me know.

bart, 2020-04-26:
Yeah, but in your equations you wrote `new_angle_roll = angle_pitch - angle_pitch ...` but it should be `new_angle_roll = angle_roll - angle_pitch ...` right? That's the only thing I changed.

gabriel-milan, 2020-04-26:
Yeah, you're absolutely correct. So sorry, I've been so distracted these days.

Anyway, thank you very much!

bart, 2020-06-29:
No problem sir. Always a pleasure to help!

closed 2020-06-29 by bart