gabriel-milan/TinyMPU6050 #19: constant AccX = 0.99 m/s²

ser, 2020-04-27:
Hi :-)

The sensor is located on my desk, completely immobile, and the value of AccX is equal constantly to 0.99 m/s² - are you able to explain why?

gabriel-milan, 2020-04-27:
Hi @ser!

This is probably a calibration issue. I've removed accel calibration on [0.5.0](https://github.com/gabriel-milan/TinyMPU6050/releases/tag/0.5.0) and didn't have time to re-add it. There are few more fixes I want to implement, I'll probably be working on it today.

Thanks for getting in touch, I'll try to work on it ASAP.

gabriel-milan, 2020-04-27:
@ser 

So, I've changed the calibration method for one I got [here](https://forum.arduino.cc/index.php?topic=619352.0). It's on the [dev branch](https://github.com/gabriel-milan/TinyMPU6050/tree/dev). As I can't test this code since I have no MPU6050 here, I'm asking you if you could please test it and give me a feedback.

As soon as this is approved, I'll post a new release. If this fails, I'll probably just copy the gyro calibration code to see if it works.

Thanks in advance!

ser, 2020-04-28:
Unfortunately nothing has changed, a stationary object gives constantly:

```
08:54:46.201 -> AccX = 0.99 m/s²
08:54:46.201 -> AccY = 0.04 m/s²
08:54:46.201 -> AccZ = -0.18 m/s²
```

ser, 2020-04-28:
But, I have just downloaded this library: https://github.com/rfetick/MPU6050_light and it shows nearly the same values. Does it mean that my sensor is broken in some way?

gabriel-milan, 2020-04-28:
Well, in that case, it's probably better for you to be sure your sensor is working well. I'll close this for now but, if you have any issues with this library, feel free to reopen it.

closed 2020-04-28 by gabriel-milan