gabriel-milan/TinyMPU6050 #20: TinyMPU return strange values if used with delay

florian-str, 2020-05-14:
I used the default "Angles Example" but added a delay of 1 sec at the end of the `void loop().` function. The start Value seems to be ok (around +-2°) but after I started to change the pitch, it starts to return weird values. e.g. even though I put it back on the flat table, it still shows the max value and reduces it really slow (around 0.5° per delay). If I remove the delay, everything is responding immediately.
I also tried to put the printing into another function called after a delay by the loop, but the error is still the same.
I've experienced this bug both on my ESP32 and Arduino UNO (maybe on other boards to but I have only these).
Else it's a great library.

gabriel-milan, 2020-05-14:
Hi @florian-str, thanks for getting in touch.

To be honest, I wouldn't call this a bug, since the angle calculations are done using the idea of integration:

```cpp
dt = (millis() - intervalStart) * 0.001;
angGyroX = wrap(angGyroX + gyroX * dt);
angGyroY = wrap(angGyroY + gyroY * dt);
angGyroZ = wrap(angGyroZ + gyroZ * dt);
```
(This snippet was extracted from [here](https://github.com/gabriel-milan/TinyMPU6050/blob/master/src/TinyMPU6050.cpp#L113))

As we know, the smaller `dt` is, the more accurate our results should be. But then, if you insert a delay between loops, `dt` grows too much, implying on awful results. Many microcontrollers like Arduino UNO are single-cored and have no workarounds on this.

On the other hand, you've mentioned an ESP32. Those guys have 2 cores and can run two parallel tasks. In that case, I would recommend that, if possible, you'd configure one of the cores for MPU6050 stuff only (or add few minor things too) and the other for your delayed loop. Please try this implementation before anything else and, if you need any help on doing so, get in touch.

I won't close this for now, since your issue is not yet fixed.

florian-str, 2020-05-15:
Thank you for your answer. I was able to make it work with the ESP32.

gabriel-milan, 2020-05-15:
Glad to hear that. If you need anything else, please let me know.

closed 2020-05-15 by gabriel-milan