gabriel-milan/TinyMPU6050 #24: Measurement delayed when when upside down

mcer12, 2020-08-30:
Hi @gabriel-milan first off, thank you so much for this, it works the best with ESP8266 from libraries I tried :) 

I don't know if this is a bug or it's a "feature" of the sensor but when I orient it upside down (MPU6050 chip facing down), the measurements are slow / delayed. It's around one second delayed. Once I reorient it back, it's fast again. This is very unfortunate as I plan on using it oriented like that.

gabriel-milan, 2020-08-31:
Hi @mcer12, thank you for appreciation!

Honestly, I probably haven't tested the library in such conditions, as I wrote it for different purposes. Can you assure it's a library issue? Have you tried it with another MPU6050/ESP8266 pair? Unfortunately, I have no access to mine, so I can't tell whether it's reproducible.

I don't think this should be regular behavior, though. There may be a bug I couldn't find yet.

mcer12, 2020-08-31:
@gabriel-milan quite honestly I wasn't able to make other libraries work (esp8266) :) 
**MPU6050_tockn** example returns the same number and doesn't react to changes.
**MPU6050_light** example crashes during calibration

I suspect this might be caused by the calculation somehow when the gyro and acceleration data are merged. I will take a look if this delay is also happening with acceleration / gyro angle only.

gabriel-milan, 2020-08-31:
> I wasn't able to make other libraries work

I can't tell for sure, but maybe it's a sensor issue. I'd be pleased if you would investigate on that, all collaboration is very welcome.

mcer12, 2020-08-31:
The sensor works perfectly OK with your library, I think it's the compatibility with ESP issue with other libs.

I viewed the values in serial plotter side by side **mpu.GetAngAccX()** and **mpu.GetAngX()** and here are the results:

GetAngX() is instant and perfectly copies and smoothens GetAngAccX() when positioned normally, on point. When positioned upside down, GetAngAccX() is instant but GetAngX() takes time to catch up. So it's absolutely a calculation issue. The graph looks like it tries to average to the wrong side... it overshoots and then slowly catches up. I can share the plotter screenshot if that helps.

mcer12, 2020-08-31:
I played a bit with the plotter and the mpu.GetAngAccX() is fast but is also a bit less precise or responsive. So I will probably redesign the project to use it in normal position. The sensor probably doesn't expect to be positioned like that.

gabriel-milan, 2020-08-31:
>  I can share the plotter screenshot if that helps.

That would be nice to see, even if you're going to redesign your project.

> The sensor probably doesn't like to be positioned like that.

I don't think that should be an issue. It's likely a library issue. Such a pity I'm not able to help on debugging/fixing these right now (and I'm not sure when I'll be able again).

closed 2020-11-06 by github-actions[bot]