gabriel-milan/TinyMPU6050 #26: Wrong angle range of values for X and Y axis, drifting Z-axis angle value

EwaldJa, 2020-12-27:
Hi again,

While using your library, running the demo sketch **ArduinoIDE_Angles_Example** on my ESP32, I noticed that the range of values for the X and Y axis is wrong : starting flat, the angle is 0. 

- when I tilt on the X-axis, the angle will go from 0 to + 90° or -90° (*depends on the tilting direction*), and then fall back to zero instead of reaching +180 or -180 and then changing sign
- same thing happens on the Y axis

I tried to attach a video of me tilting the device and recording the screen in the same time, sorry for the bad quality, it was made with my phone. Here is the video : [Wrong values MPU6050](https://drive.google.com/file/d/1FECFhYNR4pFc65QPUw_Bk1F2C-TGwEn2/view?usp=sharing) 

I have also noticed that the Z-axis angle value drifts as I tilt on X or Y axis, and never falls back to 0 when put in the same position as I used for calibration.

Regarding wrong angle range, I suspect it has something to do with the wrapping, however I have not yet looked in it, however, I do not have any idea concerning Z-axis drifting...

Best regards,

Ewald

gabriel-milan, 2020-12-30:
Hello again :)

> I suspect it has something to do with the wrapping

That's probably correct. The current method for wrapping angles was proposed on https://github.com/gabriel-milan/TinyMPU6050/pull/14 in order to fix https://github.com/gabriel-milan/TinyMPU6050/issues/6, but now it seems that it would need to change again. I still don't have my MPU6050 with me, so I'm not able to try it out for myself, unfortunately.

> I do not have any idea concerning Z-axis drifting

IMO, that may be due to your accelerometer offsets (https://github.com/gabriel-milan/TinyMPU6050/issues/25)

mcr-ksh, 2021-06-08:
same problem here. Z-axis is drifting.

EwaldJa, 2021-06-08:
I ended up not using the Z-axis. 

> same problem here. Z-axis is drifting.

After doing a lot of research, I ended up finding that you could pair the MPU6050 with a magnetometer, and using the built-in compute unit, should be able to have stable Z-axis values. I'm so stupid to not have understood this before. Because Z-axis is exactly perpendicular to the Earth, you cannot use gravity to offset accelerometer drifting away as for X and Y axis. The magnetometer is the sensor that is able to counter drifting accelerometer values. 
I know that someone retro-engineered a lot on the MPU6050, and you should be able to pair yours with a magnetometer using XCL and XDA pins.
I ended up ditching Z-axis values for my project, so I have not looked further into this matter, sorry :(

meltoner, 2022-04-19:
Hi EwaldJa, i am also using this library but i experience freeze after random times. in my project i am depend on the z degree.
Do you think this library is perhaps beta? Do you know why it was forked from https://github.com/tockn/MPU6050_tockn i mean what did it try to solve? last if you know more about this magnetometer paring please let me know. from my search i haven't found some example. Last i just can't understand how in 2022 there isn't a defacto lib for 6050 that works like a charm..

gabriel-milan, 2022-04-19:
> Do you think this library is perhaps beta?

Yes, it should be considered beta.

> what did it try to solve?

Actually, I've started this as a personal project, so I could learn from this experience. Then, I've thought about releasing it, so others could use. Also, the whole idea was to provide a reliable way of reading angles with it, but problems began to show up and, unfortunately, I'm not able to maintain this anymore.

> Last i just can't understand how in 2022 there isn't a defacto lib for 6050 that works like a charm

Totally agree with you on this

meltoner, 2022-04-19:
Thanks Gabriel for the feedback. frankly it was a relief that i found yours that is lightweight and easy to follow compared to other large ones, consuming a large percent of the memory in arduino. I am developing a return to a gps functionality where i am interested in the z axis of the mpu to find out the relative heading. ( I tracking the source code here - https://github.com/meltoner/NEM_UVRC ) 

Your library works relatively well as far as i can hope for, (when I first used i said finally a nice library ) though on more "heavy duty" testing i am faced with some random freezes in arduino after some unknown time and i am researching more about this (i was chasing the rover yesterday :) ). 

I am now using a blink led to spot when the arduino freezes, while when i comment the mpu.update() the arduino doesn't seem to freeze. I am not sure where to look perhaps some bad zero division or some non ending while loop. I think it would be great to have your library working just a bit further as it looks clean and neat compared to what i seem to find available. 

I am not sure if i should check the library, you based yours as an alternative or research further yours. I am not that confident to attempt changes alone, but perhaps if we were "joining forces" it would be easier for both of us. Let what me know what you think! :)

meltoner, 2022-04-23:
Dear Gabriel, while searching further it seem that i am not using your library. instead i am using [MPU6050_light](https://github.com/rfetick/MPU6050_light) so most probably the instability problem must not be applicable to your library. sorry for this.

meltoner, 2022-04-25:
Hi all, researching further the freezes and after trialing your library too, the freeze occurred on all cases tested. 

The issue was sourced at the i2c protocol and was solvable by setting the wiretimeout to 3 seconds as follows

`  
  Wire.begin();
  Wire.setClock(400000); 
  Wire.setWireTimeout(3000, true); 
`
  The solution discovered after many days of troubleshooting (pain. tears, etc ) was 
  - https://github.com/jrowberg/i2cdevlib/issues/519
  
  by [chrisvolto](https://github.com/chrisvolto) commented [on Dec 29, 2020](https://github.com/jrowberg/i2cdevlib/issues/519#issuecomment-752023021)
  
 over and out